PRosenb/DeepSleepScheduler #1: Does not compile for DigiSpark Pro (AtTiny167)

supuflounder, 2016-10-10:
```
Arduino: 1.6.9 (Mac OS X), Board: "Digispark Pro (Default 16 Mhz)"
...
In file included from /Users/flounder/OneDrive/Projects/XXX/V2/Sketches/Temperature/Temperature.ino:4:0:
/Users/flounder/Documents/Arduino/libraries/DeepSleepScheduler/DeepSleepScheduler.h: In member function 'Scheduler::SleepMode Scheduler::evaluateSleepModeAndEnableWdtIfRequired()':
/Users/flounder/Documents/Arduino/libraries/DeepSleepScheduler/DeepSleepScheduler.h:707:7: error: 'WDTCSR' was not declared in this scope
       WDTCSR |= (1 << WDCE) | (1 << WDIE);
       ^
/Users/flounder/Documents/Arduino/libraries/DeepSleepScheduler/DeepSleepScheduler.h:715:5: error: 'WDTCSR' was not declared in this scope
     WDTCSR |= (1 << WDCE) | (1 << WDIE);
     ^
Using library DeepSleepScheduler at version 2.0.0 in folder: /Users/flounder/Documents/Arduino/libraries/DeepSleepScheduler 
...
exit status 1
Error compiling for board Digispark Pro (Default 16 Mhz).
```

Note that this compiles for the Arduino UNO and the Adafruit 32U4 Feathers.  The code appears to be consistent with the code shown on page 55 of the AtTiny167 data sheet, except for the name of the register (WDTCR) and the C-language interface (which I presume is a difference between the Atmel IDE and the Aduino IDE).

We are on a very short fuse to deliver a prototype, so an immediate reply would be greatly appreciated.
     joe

PRosenb, 2016-10-10:
Hi Joe,
Do you know what the change needs to be to support both CPU's? I'm happy to add ifdef to support them both. Perfect would be if you could do a pull request with it.
Cheers,
Pete

PRosenb, 2016-10-11:
Hi Joe,
I added `#ifdef` for the AtTiny CPUs and it compiles now.
Unfortunately, I don't have such a CPU to do tests. Could you do that?
The change is on the following branch:
https://github.com/PRosenb/DeepSleepScheduler/tree/supportAtTiny167

Thanks,
Pete

supuflounder, 2016-10-12:
> Hi Joe,
> I added `#ifdef` for the AtTiny CPUs and it compiles now.
> Unfortunately, I don't have such a CPU to do tests. Could you do that?
> The change is on the following branch: [...]
Thanks.  I will download it sometime later tonight, after I have enough
sleep that I'm not falling facedown into the keyboard (you know your hours
are too long when people start asking "Why on earth did you have your face
tatooed with RTYFGHCVBN and set it up so you can only read it in a
mirror?")

PRosenb, 2016-10-12:
Hi Joe,
How are you going with the changes?
Cheers,
Pete

supuflounder, 2016-10-12:
Nothing. I am finally out of bed after 24 hours of misery, and have been
at the computer less than ten minutes.
       joe

PRosenb, 2016-10-13:
Is it working now?

closed 2016-10-25 by PRosenb