PRosenb/DeepSleepScheduler #2: Compiler warnings

patman15, 2017-04-29:
Setting compiler warnings to "more" on ATmega328 causes the following warnings.
Can they be fixed to avoid covering own issues?

Thanks for the cool library!
```
/Arduino/libraries/DeepSleepScheduler/DeepSleepScheduler.h: In member function 'void Scheduler::removeCallbacks(void (*)())':
/Arduino/libraries/DeepSleepScheduler/DeepSleepScheduler.h:511:18: warning: deleting object of abstract class type 'Scheduler::Task' which has non-virtual destructor will cause undefined behaviour [-Wdelete-non-virtual-dtor]
           delete taskToDelete;
                  ^
/Arduino/libraries/DeepSleepScheduler/DeepSleepScheduler.h: In member function 'void Scheduler::removeCallbacks(Runnable*)':
/Arduino/libraries/DeepSleepScheduler/DeepSleepScheduler.h:536:18: warning: deleting object of abstract class type 'Scheduler::Task' which has non-virtual destructor will cause undefined behaviour [-Wdelete-non-virtual-dtor]
           delete taskToDelete;
                  ^
/Arduino/libraries/DeepSleepScheduler/DeepSleepScheduler.h: In member function 'void Scheduler::execute()':
/Arduino/libraries/DeepSleepScheduler/DeepSleepScheduler.h:615:16: warning: deleting object of abstract class type 'Scheduler::Task' which has non-virtual destructor will cause undefined behaviour [-Wdelete-non-virtual-dtor]
         delete current;
                ^
/Arduino/libraries/DeepSleepScheduler/DeepSleepScheduler.h: In member function 'long unsigned int Scheduler::getScheduleTimeOfCurrentTask() const':
/Arduino/libraries/DeepSleepScheduler/DeepSleepScheduler.h:494:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
```

PRosenb, 2017-04-29:
Hi Patman15, thanks for reporting these compiler warnings. Can you please explain in more details what compiler you use and how you run it so that I can reproduce the warnings?

patman15, 2017-04-30:
Hi PRosenb,

sorry for that. I'm using the current [Arduino IDE](https://www.arduino.cc/en/Main/Software) 1.8.2 under Ubuntu 17.04 64 Bit. To make it simple, I attached the [compile log](https://github.com/PRosenb/DeepSleepScheduler/files/966570/compile_log.txt) of the _BlinkRunnable_ example. Between the _default_ and the _more_ setting the IDE adds the _-Wall_ option to the compiler call as far as I can see.

PRosenb, 2017-04-30:
Thanks for the details. I saw the warnings on Mac and updated the code on master. It would be great to verify on your side as well if the warnings are gone.

patman15, 2017-04-30:
Thanks for the quick fix. I'm currently away from my development system. I'll test it and give you feedback next week.

patman15, 2017-05-07:
Compiles now without any warnings. Thanks!

closed 2017-05-07 by patman15