PRosenb/DeepSleepScheduler #3: swap SLEEP_MODE at run time?

Jim4Beam, 2017-07-26:
Hi,
sorry not an Issue, but a question, will it be possible to change the SLEEP_MODE at runtime? Or what needs to be changed to have the option to change it at runtime? Will it be as simple as changing SLEEP_MODE from a #define to a bool or byte variable? 

Background: I will use it for a device that will dim an LED (PWM, therefore no deep sleep) for some hours a day but could deep sleep most time of the day in the lowest sleep mode to save power.

Thanks for considering and sorry for the "issue" (I do not know how to ask questions at GitHub)
Best Regards

PRosenb, 2017-07-27:
Hi Jim,
Nice to hear that you wanna use my library. No worries, you can also ask questions here.

You can get this behaviour with the two methods
`scheduler.acquireNoDeepSleepLock()`
and
`scheduler.releaseNoDeepSleepLock()`

The first one prevents that the CPU is put to sleep and the second one allows it again. Please make sure you call them the same amount of times.

Also see their description:
https://github.com/PRosenb/DeepSleepScheduler#methods
And this example:
https://github.com/PRosenb/DeepSleepScheduler/blob/master/examples/ShowSleep/ShowSleep.ino

I hope that works for you and good luck with your project.

Cheers,
Pete

Jim4Beam, 2017-07-30:
Hi Pete,

thank you very much for the fast answer. Great that you have a method to 
switch between deep sleep an no deep sleep at run time. Will the CPU be 
in Idel Mode if |scheduler.acquireNoDeepSleepLock()| was called or 
continue running full speed and power?

Best regards
Jim

PRosenb, 2017-07-30:
Hi Jim,
Yes, the CPU will be in Idle mode while you don't schedule tasks. It is woken up every millisecond though by the timer tick and then goes back to Idle.
Cheers,
Pete

closed 2018-01-22 by PRosenb