PRosenb/DeepSleepScheduler #4: Error printing to Serial Monitor

guimaraes, 2018-06-24:
When my project enters DeepSleep and restarts it prints on the Serial Monitor a sequence of strange characters.

The project is being powered by batteries and needs to get into DeepSleep, to save batteries and this is being done and doing very well. But, I need the system to turn on every 30 seconds, light a led on pin 13 and go back to the DeepSleep function.

I tried to use another library, DeepSleepScheduler, and this one believes that it is doing what it is programmed to do, but when it prints in Serial, it shows a series of strange characters, as shown in the attached video.

Here is the code I am using for testing:

`#include` <DeepSleepScheduler.h>
#define BUTTON 2
#define BLUE_LED 6
#define DEEP_SLEEP_DELAY 2000

int AN[] = {A1, A2, A3, A4, A5};

void setup() {
  for(int i = 0; i < 14; i++){//Coloca todas as portas digitais em LOW
    digitalWrite(i,LOW);
  }
  for(int i = 1; i < 5; i++){//Coloca todas as portas analogicas em LOW
    digitalWrite(AN[i], LOW);
  }
  pinMode(BLUE_LED, OUTPUT);
  Serial.begin(1200);
  scheduler.schedule(executFuncoes);
}

void loop() {
  scheduler.execute();
}

void executFuncoes(){
  lerBlueLeds();
  Serial.println("Teste! ");
  scheduler.scheduleDelayed(executFuncoes, 5000);
}

void lerBlueLeds(){
  for(int i=0; i<=50; i++){
    digitalWrite(BLUE_LED,HIGH);
    delay(30);
    digitalWrite(BLUE_LED,LOW);
    delay(30);
  }  
}`

PRosenb, 2018-06-24:
Hi Neto, thanks for considering my library. You need to do the library related `#define` options **before** the `#include`, otherwise they have no effect.
I will rename DEEP_SLEEP_DELAY to SLEEP_DELAY so I put both below to make sure it still works afterwards.

It should read
```c++
#define DEEP_SLEEP_DELAY 2000
#define SLEEP_DELAY 2000
#include <DeepSleepScheduler.h>
#define BUTTON 2
#define BLUE_LED 6
...
```
Please let me know if that solves your issue, cheers.

guimaraes, 2018-06-25:
Muito obrigado, deu certo!
Eu agradeço demais!

PRosenb, 2018-06-25:
I am happy to hear that it works, enjoy.

closed 2018-06-25 by PRosenb