PRosenb/DeepSleepScheduler #5: External Interrupt

guimaraes, 2018-06-26:
Pete, good night!

The project I'm developing is a solution that needs to be triggered by external interruption and need to send a die every 30 seconds and go back to deep sleep.

As the solution was before using his library I was able to make use of the external interrupt, but I could not get him to wake up every 30 seconds.

Do you have any way to solve this problem?

guimaraes, 2018-06-26:
Thank you again!
I decided!

PRosenb, 2018-06-26:
Interrupts work as well, just the timing can vary so you might have to adjust the schedule time.

The following script switches the built in LED on/off every 30 seconds and prints to the console if you touch PIN 2. Should you touch it quickly you might want to do scheduler.scheduleDelayed() from the interrupt to prevent multiple executions.

```c++
#define DEEP_SLEEP_DELAY 2000
#define SLEEP_DELAY 2000
#include <DeepSleepScheduler.h>

#define INTERRUPT_PIN 2

void ledOn() {
  digitalWrite(LED_BUILTIN, HIGH);
  scheduler.scheduleDelayed(ledOff, 30000);
}

void ledOff() {
  digitalWrite(LED_BUILTIN, LOW);
  scheduler.scheduleDelayed(ledOn, 30000);
}

void interruptPin() {
  Serial.println("interrupt happened");
  attachInterrupt(digitalPinToInterrupt(INTERRUPT_PIN), isrInterruptPin, FALLING);
}

void isrInterruptPin() {
  scheduler.schedule(interruptPin);
  // detach interrupt to prevent executing it multiple
  // time when touching more more than once.
  detachInterrupt(digitalPinToInterrupt(INTERRUPT_PIN));
}

void setup() {
  Serial.begin(1200);
  
  pinMode(INTERRUPT_PIN, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(INTERRUPT_PIN), isrInterruptPin, FALLING);
  
  pinMode(LED_BUILTIN, OUTPUT);
  
  scheduler.schedule(ledOn);
}

void loop() {
  scheduler.execute();
}
```

PRosenb, 2018-06-30:
Hi @guimaraes does it work for you like that?

guimaraes, 2018-07-12:
Pete, good afternoon!

Now I have a slightly bigger problem!

My project needs to flash the green LED every 30 seconds to indicate that the equipment is on, and when it has an external interruption (when the button is pressed) it must send a code with the id of the equipment by radio using the module HC12 RF (Code model: 1#BT0001&op=1) and to reach the central and return it takes a time that can take from 3 to 5 seconds and to wait for this time I am using a loop of 150 reps, to wait for the id to return from the central. The Central returns me the same added id of the character "c", to indicate that this is the id coming from the central. If this feedback is recognized by the device, the system leaves the loop, and enters the method that checks the state of the battery, if the battery charge is below 3.5v, it enters the conditional and sends to the central one the incremented with "&op=5" to indicate that it is a battery occurrence, after checking the battery, it enters a loop of 50 repetitions and the HC12 RF 433MhZ module is turned off.

When I add the code that you posted with my code, it blocks the microcontroller I'm using, ATmega 328p.

Thank you very much for your attention and I would like to ask for your help in solving this problem.

My WhatsApp is: +55 62 99553 1804

This is my original code:

```
#include <SoftwareSerial.h>

#include <avr/power.h>
#include <avr/sleep.h>
#include <avr/io.h>
#include <avr/interrupt.h>
#include <avr/wdt.h>

SoftwareSerial mS(4, 3);

#define BUTTON        2

boolean buttonPress = false; 

#define RED_LED       5
#define HC12_START   12

int AN[] = {A1, A2, A3, A4, A5};

String receiveCentral = "";

void setup() {
  mS.begin(1200);
  Serial.begin(1200);
  
  for(int i = 0; i < 14; i++){
    digitalWrite(i,LOW);
  }
  for(int i = 1; i < 5; i++){
    digitalWrite(AN[i], LOW);
  }

  pinMode(BUTTON           , INPUT_PULLUP);
  pinMode(LED_BUILTIN      , OUTPUT);
  pinMode(RED_LED          , OUTPUT);

  pinMode(HC12_START       , OUTPUT);
  digitalWrite(HC12_START  , HIGH);
}

void loop(){
  pushButtonPressed();
  Serial.println("Desligar processador!");
  sleepNow();
}

/**
 * Metodo utilizado quando o botão é pressionado
 */
void pushButtonPressed(){
  if (digitalRead(BUTTON)==0){
    digitalWrite(HC12_START,LOW);//Liga o modulo HC12 RF 433
    
    String ID_DEVICE = "1#BT0001";//Passa o ID do dispositivo para a variavel
    String idShipping = ID_DEVICE+"&op=1";//Incrementa o ID do dispositivo com o tipo de ocorrencia
    
//    readLedRed(100);
    
    while(receiveCentral != ID_DEVICE){//Laço responsavel por enviar dado para central até receber resposta
      mS.print(idShipping);//Envia dados para central
      validateCentral(); //Metodo responsavel por validar dados vindo da central, passa o ID da central como parametro
    }
    
    receiveCentral = "";
    batteryCurrent(ID_DEVICE);//Verifica o
    readLedRed(50);
    
    digitalWrite(HC12_START,HIGH);//Desliga o modulo HC12
  } 
}

void readLedRed(int numberLoop){
   for(int i=0; i<=numberLoop; i++){
    digitalWrite(RED_LED,HIGH);
    delay(30);
    digitalWrite(RED_LED,LOW);
    delay(30);
  }  
}

/*
 * Valida dados vindo da central
 */
void validateCentral(){
  readLedRed(150);//Para aguardar os dados vindos da central, é passado um laço de 20 repetições que permitem piscar o LED Vermelho
  if(mS.available() > 1){//Caso a serial tenha recebido uma informação da central entra no metodo plantData()
    plantData();
  }
}

/*
 * Verifica os dados vindos da central pertencem ao dispositivo
 */
void plantData(){
  char c;//Caracter utilizado para tratar dados vindo da central
  String validator = "";//Recebe o valor vindo da central sem esta tratado
  receiveCentral = "";//Variavel global, recebe os dados vindo da central tratado e compara com o ID do dispositivo
  validator = mS.readString();//Passa dados lidos na porta SERIAL
  for(int i = 0; i < 6; i++){//Laço de repetição utilizado para tratar dado vindo da central. 
    c = validator.charAt(i);
    receiveCentral += c;
  }
  receiveCentral = "1#"+receiveCentral;//incrementa id do alarme do dispositivo 
}

void sleepNow(void){
  attachInterrupt(digitalPinToInterrupt(BUTTON),funcInterrupcao,LOW);
  delay(100);
  set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  sleep_enable();
  sleep_mode();
  sleep_disable();
  digitalWrite(13,LOW);
}

void funcInterrupcao(void){
  detachInterrupt(BUTTON);
}

void batteryCurrent(String idDevice){
  Serial.print("- Bateria: ");
  Serial.println(((analogRead(A0))*0.003225806452*4.9000));
  if(((analogRead(A0))*0.003225806452*4.9000)<=3.50){
    idDevice = idDevice+"&op=5";
    mS.print(idDevice);
    Serial.println(idDevice);
    readLedRed(10);
  }
}
```

guimaraes, 2018-07-12:
[image]

PRosenb, 2018-08-29:
@guimaraes Could you resolve your issue?

closed 2019-01-24 by PRosenb