PRosenb/DeepSleepScheduler #6: add scheduleOnce()

paulo-raca, 2019-01-03:
First of all, thanks for this great library, I've been using it on most of my Arduino projects :)

One thing I've been doing again-and-again is to schedule a task to process some interruption event. But if the interruption happens several times before the task has a chance to run, it just needs to run once.

So far I've been using this to ensure tasks are only schedule once:
``
scheduler.removeCallbacks(task);
scheduler.schedule(task);
```

I suggest we add a new function `scheduleOnce()`, which avoids this issue.
Makes sense? What do you think?

PRosenb, 2019-01-07:
Hi @paulo-raca , great to hear do you like my library.
I generally try to keep it as simple, small and easy to use as possible.

I see your point in adding the function `scheduleOnce()` so that users of the library can write less code.
On the other hand may it not be very clear what `once` means because it could also mean once per call.. An other small drawback is, that it adds more methods to the API. The users may then ask themselves which method to use.

Given the pros and cons I think it's better to leave it as is. What do you think?

paulo-raca, 2019-01-07:
It's not just about lazyness, but also about correctness and performance:

- Correctness: Even if a task is triggered many times, it doesn't mean it needs to scheduled many times, and it is probably better if it doesn't. E.g., my debouncer implementation: https://github.com/paulo-raca/YetAnotherArduinoDebounceLibrary/blob/master/examples/Async/Async.ino#L43

- Performance: Doing both in one operation will possibly be a bit faster, which is important in an ISR -- See below.

----------------

Let me give you context on what I am doing:

I have a project called [Expanduino](https://github.com/Expanduino/Expanduino-Arduino), in which a linux host talks to an Arduino over I2c and sends commands to several of it's peripherals: LCD, serial ports, gpios, etc.

Each of these commands are sent as an `I2C write` with the commands parameters, quickly followed by an `I2C read` with a response.

One of these peripherals is an HD44780 character display -- Sending data to the display requires bitbanging the signals and waiting the appropriate delays.

Depending on the operation, it takes a few milliseconds and therefore cannot be done inside I2C interruption. 
Instead, I put the commands in a queue and use DeepSleepScheduler to transfer them asynchronously (see https://github.com/paulo-raca/Arduino_AsyncLiquidCrystal).

After queueing, I need to trigger the executing with `schedule()` -- Unless the LCD transfer is alteady scheduled, in which case I do not want it to be scheduled twice -- That's why I want `scheduleOnce()`

By itself this wouldn't be a big problem, but since in this particular project the I2C response is read shortly after the command is received, I have very few microseconds to process the request interruption, and the calls to `scheduler.removeCallbacks()` + `scheduler.schedule()` were barely over my time time slot. 

Efficiently Merging them into one function would probably have made the timing issue go away :grimacing: 

In the mean time, I've settle for another good-enough solution:
LCD function always re-scheduling itself, even if no data is present, so that I don't need to worry about scheduling inside the I2C interrupt.

PRosenb, 2019-01-07:
Thanks for your detailed reply. I like your arguments to add this function, especially the performance gain we can hopefully achieve.
You basically say `scheduleOnce()` should go through the list of all tasks once only and check at the same time if the new task is in already and where to put it in. Correct?

paulo-raca, 2019-01-14:
Yes, that sounds right.

Thank you for looking into this :)

PRosenb, 2019-01-27:
Hi @paulo-raca, I implemented a draft version of `scheduleOnce()` on the branch [ScheduleOnce](https://github.com/PRosenb/DeepSleepScheduler/tree/scheduleOnce).
Please have a look to see how works in your application.

paulo-raca, 2019-01-27:
Thank you very much.

I glanced on it and it looks good.
I'll take a better look tomorrow and give it a try

Thank you : 💓

PRosenb, 2019-02-08:
How did you go @paulo-raca with using `scheduleOnce()`?

closed 2019-04-20 by PRosenb