PRosenb/DeepSleepScheduler #9: ESP8266 resetting

kulwinderjit, 2019-05-21:
Hi,

my esp8266 keep resetting if i call scheduler.execute(); in loop, if i remove this line then it stops resetting.
ets Jan  8 2013,rst cause:4, boot mode:(3,6)

Here is how call back is registered
scheduler.scheduleDelayed(initializeModem, 60000);

Callback code
void initializeModem(){
  SerialMon.println(F("Initializing modem..."));
}

loop method
void loop(void)
{
  httpServer.handleClient();
  MDNS.update();
  handleMovement();
  handleCall();
  scheduler.execute();
}

PRosenb, 2019-05-21:
Hi,

Thanks for your message. The idea is to only have `scheduler.execute()` in your `loop()` method like this:
```C++
void loop() {
  scheduler.execute();
}
```

Please remove all other code in `loop()`. Does it still reset the esp8266?

You can move the code you have in `loop()` to callback methods like your `initializeModem()` and schedule it on regular basis.

kulwinderjit, 2019-05-22:
It does not work with simple code. I am using nodemcu dev board featuring esp12e. I have checked that delay of up to 8 seconds work but not 9+ seconds.

`#include <DeepSleepScheduler.h>

void setup() {

  Serial.begin(115200);

  scheduler.scheduleDelayed(initializeModem, 9000);

}

void loop() {

  scheduler.execute();

}

void initializeModem(){

  Serial.println();

  Serial.println("Initializing modem...");

}`

`ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vac02aff5
~ld
`

PRosenb, 2019-05-25:
Hi @kulwinderjit ,
Thanks for testing it out. Unfortunately, I don't have an ESP8266 here at the moment. Could you check if the following workaround solves the issue?

```c++
#include "DeepSleepScheduler.h"

void periodicExecution() {
  scheduler.scheduleDelayed(periodicExecution, 5000);
}

void setup() {
  scheduler.schedule(periodicExecution);
}

void loop() {
  scheduler.execute();
}
```

kulwinderjit, 2019-05-26:
It works with delay 5000 but not with 9000 and more.

kulwinderjit, 2019-05-26:
Updated library to 3.2. It does not work 

ts Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

PRosenb, 2019-05-26:
Hi @kulwinderjit 
Could you try my code above? Does it work with it?
Thanks,
Pete

kulwinderjit, 2019-05-26:
yes i have tried your code it worked and then i changed 5000 to 9000 and it started resetting again. Then i saw there is library update 3.2 after updating it stopped working with 5000 and even 1000.

PRosenb, 2019-05-26:
I see, thanks for testing it. The version 3.2 is not yet released. Please continue to use 3.1 for now and I will look into it when I get an ESP8266 again.

kulwinderjit, 2019-05-26:
I see. I got 3.2 update on platformio but not in arduino ide.

PRosenb, 2019-05-26:
Interesting.. So platformio seems not to honour the tag.
I hope you're not blocked any more with 3.1 and the workaround above.

kulwinderjit, 2019-05-26:
> yes i have tried your code it worked and then i changed 5000 to 9000 and it started resetting again
It does not work with interval 9 sec or more. I have requirement of around 60 second timeout. Your code worked because you put 5000 ms interval which works.

PRosenb, 2019-05-29:
Hi @kulwinderjit,
I got an ESP8266 and updated DeepSleepScheduler on master. Does it work well for you now?

closed 2019-06-14 by PRosenb