PRosenb/DeepSleepScheduler #10: Documentation on used sleep mode

patman15, 2019-12-25:
I was playing around with the AWAKE_INDICATION_PIN and wondered why my program never turned the LED of while the example (ShowSleep) did. The tiny but important difference was the sleep time. After quite some debugging I found:
`DeepSleepScheduler_avr_implementation:132: // use SLEEP_MODE_IDLE for values less then SLEEP_TIME_1S`
May I ask if you could add a hint to the documentation of "AVR specific options", #define SLEEP_MODE or the implementation notes or the example? Maybe it saves some others time for debugging.
By the way is there a specific reason why 1s is used and not e.g. 100ms? Just wondering.

PRosenb, 2019-12-26:
Hi @patman15 , thanks for your input. That's a very good idea to mention it on the example, I just added a comment there.
It's a good question why to use 1s. I don't remember a hard reason to use 1s. What I noticed is, that the sleep duration of the watchdog is less accurate then idle mode of the CPU.
We could check it out if lowering it to e.g. 100ms does not cause any issues.

patman15, 2019-12-26:
Hi @PRosenb incredible response time :-)
Well actually I started from polling the serial port every ~5ms and wondered why it's not going to deep sleep. I now switch to IRQ driven operation and I'm using SLEEP_MODE_ADC because that allows the UART to not loose data on wakeup. (Higher modes (SLEEP_MODE_STANDBY) didn't work without lost characters). In the end lowering the value wouldn't have helped my issue.

I still thought a bit about lowering the value regarding accuracy. Basically, due to various dependencies (voltage, temperature) you cannot predict it very well. So the question is from which value on people do not expect exact timing any more. 1s is definitely on the safe side. I personally wouldn't go lower than 100ms (->SLEEP_TIME_120MS) because you might want to have e.g. some precise 50Hz task.

Finally, thanks for adding the hint, most probably that hinders further issues already. ;-)

PRosenb, 2019-12-26:
Hi @patman15 , thanks for your quick response too :-).
Interesting.. That's cool are you using my library for that purpose. I hope it helps and works well in your context.
Yes I agree, an important consideration is about expected accuracy. An option would also be to make it configurable with a `#define` and set the default to e.g. `SLEEP_TIME_120MS + BUFFER_TIME` or even leave it at `SLEEP_TIME_1S` to have no change by default.

patman15, 2019-12-28:
Following "never change a running system" I would keep the value as there is no specific reason to do a change. The documentation helps a lot and at maximum I would make it changeable with a define similar to SLEEP_MODE.

PRosenb, 2019-12-28:
Added define option `MIN_WAIT_TIME_FOR_SLEEP` to allow to change it and set default to existing value.

closed 2019-12-28 by PRosenb