PRosenb/DeepSleepScheduler #11: Daily task

tech-freak, 2020-06-13:
hi there, thank you for making this awesome library 
i have used in multiple fun projects but i am working on battery projected and i need to save some juice xD 

i want to know that is it possible to make NTP based DeepSleepScheduler ?

like i can define time 11:00 AM and 11:00 PM of task then esp boots , fetch current time from NTP and see if its time to do task or not if its  time then do the task or go back to sleep and increase boot count 
we can use boot count to avoid connecting NTP again and again

thank you once again :)

PRosenb, 2020-06-13:
Hi @tech-freak, thanks, I'm happy to hear that.

I guess you think on using `DeepSleep`, wake up at certain times and do some actions.
My application had that requirement too and I investigated how I could add such a behaviour to `DeepSleepScheduler`.
What makes it complicated is, that only memory of type `RTC_DATA_ATTR` survives `DeepSleep`. So I first thought to put the task list in there. But the problem then is, that the tasks themselves are still in main memory and main memory is cleared on `DeepSleep`. Eventhough it might be possible, I think it would make `DeepSleepScheduler` too complicated. 
I suggest to use `DeepSleep` manually instead and when the CPU starts again, use `DeepSleepScheduler` until it starts `DeepSleep` again.

With the following command, you can `DeepSleep` for a certain amount of milliseconds.
`esp_sleep_enable_timer_wakeup(durationMs * 1000L);`

I used it to wake up every 10 minutes at 10 past, 20 past and so on with calculating `durationMs` accordingly.
You can also do much longer times as [documented here](https://github.com/espressif/esp-idf/blob/045aaf6fb0e1c7bc121088a6457873ddf86badee/components/soc/esp32/rtc_time.c#L132).

To count boot times you can put `bootCount` in `RTC_DATA_ATTR` to survive `DeepDleep` like this.
`RTC_DATA_ATTR unsigned long bootCount = 0;`
It only works when you put it on top level, not in a class.

Is such a behaviour an option for you?

tech-freak, 2020-06-14:
yes i know about that i am talking about NTP specifically , its a pet feeder so i need exact time

there is a library for that too but not working now
[ESPDailyTaskNTP](https://github.com/SensorsIot/ESPDailyTaskNTP)

PRosenb, 2020-06-14:
I see.
For exact timing I did `DeepSleep` a bit less long than required, then calculated the remaining time and scheduled a task with `DeepSleepSchedler` after the remaining amount of milliseconds. Such a behaviour I encapsulated within the class `DeepSleepController` so that the rest of the code is as less affected as possible.
An option would be to implement such a behaviour in a library on top of `DeepSleepScheduler`.

closed 2020-06-16 by PRosenb