PRosenb/DeepSleepScheduler #14: MIN_WAIT_TIME_FOR_SLEEP

patman15, 2021-06-12:
> AVR specific options
> 
>     #define MIN_WAIT_TIME_FOR_SLEEP: Specify the minimum wait time (until the next task will be executed) to put the CPU in sleep mode. Default is 1 second.

Does this exist? I couldn't find any references to this option in the code.

PRosenb, 2021-06-12:
Hi @patman15 , nice are you looking at my library. You can find this definition in the AVR specific header file [DeepSleepScheduler_avr_definition.h](https://github.com/PRosenb/DeepSleepScheduler/blob/00090c6d9f881c507fc2eb7c597f9df6c37ddcbb/DeepSleepScheduler_avr_definition.h#L15).
Cheers, Pete

closed 2021-06-12 by PRosenb

patman15, 2021-06-13:
@PRosenb sorry for my confusion yesterday, I now found out why it was so simple for you and I did not see it. You are obviously right with this
> You can find this definition in the AVR specific header file [DeepSleepScheduler_avr_definition.h](https://github.com/PRosenb/DeepSleepScheduler/blob/00090c6d9f881c507fc2eb7c597f9df6c37ddcbb/DeepSleepScheduler_avr_definition.h#L15).

I still couldn't find it locally. The reason is that I installed the library from the Arduino IDE. Library manager and the install directory both mention version 3.2.0 which is the same as the master branch. In my opinion the confusion comes from the installation way. The IDE takes the [3.2.0 release](https://github.com/PRosenb/DeepSleepScheduler/releases/tag/v3.2.0) which
> PRosenb released this on 14 Jun 2019 · 8 commits to master since this release

was released earlier and [misses the commits](https://github.com/PRosenb/DeepSleepScheduler/compare/v3.2.0...master) for the mentioned option. Since both have the same version number I did not notice.
Best regards,
Patrick

PRosenb, 2021-06-13:
Hi @patman15 , a very good point, even I didn't notice that it was not released yet. I just incremented the version and created a new release 3.3.0 that should be available in the Arduino IDE soon.